fliphess/chuangmi-720p-hack #6: Night mode doesn't switch on the IR LEDs

jymbob, 2018-09-25:
auto-night-mode should switch on the IR LEDs (at a user-defined level?) but it doesn't, because the ir_sample daemon has locked access to /dev/ftpwmtmr010

If you kill the daemon, you can switch the LEDs on, but the daemon is what's checking whether it's dark enough for night mode

steps to reproduce:
ssh into camera
run `chuangmi_ctrl IRLED 255`
expected result: IR LEDs turn on
actual result: error: Failed to open /dev/ftpwmtmr010

fliphess, 2018-10-11:
What option would you prefer @jymbob ? 

The auto night detection is done by a `ir_sample` binary which I don't have the source code for in the sample libraries (like many of the mii stuffs that is used in the original firmware). 

I think they never got as far as using the pwm ir led device (`/dev/ftpwmtmr010`), while they actually did lock it, because i can't find a way to make it enable the IR leds. 

I prefer not to use it anymore and implement a similar mechanism for auto switching day and night mode using probably the same determination mechanism but in a way that it provides more control. This is for the long term ofc, because it will take some time to write such a utility.

For now I'm thinking of removing the auto night mode feature as it makes too much noise for my personal preference and as it's not fully configurable it does not work very precise. 

Second: it can cause wear-out on the camera board that does the night/day switching mechanically and afaik has a maximum amount of switches.

the other option is leaving it there, making it configurable, but block all night_mode and ir_cut/ir_led switching through web or mqtt when it is enabled. 

I don't like current way night mode switching is handled very much, so taking it out would be the easiest. Do you consider the auto night mode by `ir_sample` useful?  In other words: Are there some arguments to consider keeping it and block ir_* changes when it is on?

jymbob, 2018-10-16:
Hm. Good questions. I have a use case where it's useful, and another where it's not (in its current form).

I think a good workaround (for now) would be to remove it and therefore make it so the manual `nightmode` command works as expected (including switching on the IR LEDs). 

A future version where the sample frequency and activation level can be modified would be brilliant.

Just to clarify what I wrote earlier: if you `killall ir_sample` then the ir_led device is available again.

fliphess, 2018-10-21:
TLDR: I've made some changes: `ir_sample` is now only started at boot when `AUTO_NIGHT_MODE=1` is set in `config.cfg`. 

If the `AUTO_NIGHT_MODE` variable is set to 1, the `ir_sample` daemon is started accordingly and is the only thing able to change the night mode, ir leds etc.

The functions for sh in `functions.sh` and the new php libraries, will not be able to change the  ir leds, ir cut, night mode etc. but return an error that it's not possible due to auto-night-mode instead. 

This way you won't get the error as it's either change the settings manually or otherwise let `ir_sample` do its "job".

That way the `ir_sample` daemon is not biting me while other people are able to use it in case they think it's useful until I've created a replacement in the far future.

closed 2018-10-21 by fliphess