fliphess/chuangmi-720p-hack #7: Enable audio on RTSP stream

jymbob, 2018-09-25:
non-trivial, but audio can be captured by the microphone and output into the stream along with video.

fliphess, 2018-09-25:
Ah Nice! 

I'm currently looking into how to use the current rtsp daemon or whether i should replace it with a more functional one that supports capturing pictures and combining audio as well. 

More on that later this month :)

jymbob, 2018-09-25:
Great stuff! There are a few people who were looking at doing this with the mijia 1080p - discussions on https://gitter.im/fang-hacks/Mijia-1080P - scroll back to June, look for sangregoriopaulo

fliphess, 2018-09-25:
I've been playing around with the samples available in the sdk and it looks good: I'm able to capture images and video, play audio, record audio and some other functionalities that need lots of work but are cool enough going down the rabbit hole for :)

The native rtsp code grain media provided is very usable.
I'm not a good C programmer, so it will take some time, but the examples are easy enough to get something started :)

See if I can enable the audio, as the code already provides it.....

fliphess, 2018-12-11:
To give an update on this issue: I'm currently working on reintroducing snapshot and motion detection without using the cloud suite. As this is complex enough for me, as I'm still learning how to do C/CPP, adding audio will take some more time. 

I expect this to take at least 6 months, but maybe even longer. 

Any help is appreciated, so if any high skilled multimedia programmer reads this: Yes please! 🥇

fliphess, 2019-01-27:
I'm moving all bugs and feature requests to their own project tracker. I feel I'm spending too much time on debugging configuration issues that are documented in the wiki so to focus on new features I'm closing this issue. 

I encourage everyone that runs into bugs to debug them theirselves and create a PR to fix the issue upstream.

closed 2019-01-27 by fliphess