fliphess/chuangmi-720p-hack #14: Error when connect to web page

chris777c, 2018-10-07:
Hello,

I have one problem when I connect to webserver http://ip_camera .I go to php page configuration with this message on first line and next  table of php configuration : 

> Warning: include_once(/tmp/sd/firmware/www/include/environment.php): failed to open stream: No such file or directory in /tmp/sd/firmware/www/index.php on line 5
> 
> Warning: include_once(): Failed opening '/tmp/sd/firmware/www/include/environment.php' for inclusion (include_path='.:') in /tmp/sd/firmware/www/index.php on line 5

fliphess, 2018-10-08:
Ah I'm sorry, some of the commits of my other branch came through, I'll push a fix in the evening.....

fliphess, 2018-10-08:
I've created a new build that should fix this issue. Thanks a lot for letting me know! :)

closed 2018-10-08 by fliphess