fliphess/chuangmi-720p-hack #16: I can't copy .env file

chris777c, 2018-10-09:
Hello,

When i try copy your release 0.953, i have trouble with .env file because is the symbolic.
[image]

so I have trouble in web server :+1: 

> 
Fatal error: Uncaught Dotenv\Exception\InvalidPathException: Unable to read the environment file at /tmp/sd/firmware/www/.env. in /tmp/sd/firmware/www/vendor/vlucas/phpdotenv/src/Loader.php:105 Stack trace: #0 /tmp/sd/firmware/www/vendor/vlucas/phpdotenv/src/Loader.php(82): Dotenv\Loader->ensureFileIsReadable() #1 /tmp/sd/firmware/www/vendor/vlucas/phpdotenv/src/Dotenv.php(106): Dotenv\Loader->load() #2 /tmp/sd/firmware/www/vendor/vlucas/phpdotenv/src/Dotenv.php(50): Dotenv\Dotenv->loadData() #3 /tmp/sd/firmware/www/include/app.php(3): Dotenv\Dotenv->load() #4 /tmp/sd/firmware/www/index.php(6): include_once('/tmp/sd/firmwar...') #5 {main} thrown in /tmp/sd/firmware/www/vendor/vlucas/phpdotenv/src/Loader.php on line 105

fliphess, 2018-10-10:
Ai! The `.env` is a symlink to `/tmp/sd/config.cfg`, but I realise symlinks will not work on the fat32 filesystem so this is causing issues. 

During my test the copy action created the link as a file rather than a symlink, which is why I didn't notice myself, but apparently on other systems the `.env` file is skipped with an error resulting in this error......

As a quickfix/workaround, can you check if it works if you copy the `config.cfg` to `www/.env`?

On Mac/Linux:
```
cd <sdmount>
cp config.cfg firmware/www/.env
```

On the camera over telnet/ssh:
```
cp /tmp/sd/config.cfg /tmp/sd/firmware/www/.env 
```

On Windows: 
Copy the `config.cfg` file to into the www directory on the sdcard (in the firmware/ dir) and rename it to `.env`)

In the next release (will be there later this week as I'm still working on some small web thingies i want to finish first), I'll completely remove the symlink and use the full path to the file instead.

chris777c, 2018-10-15:
Hello,

Thank you, it's work for me

fliphess, 2018-10-15:
Cool! Thanks for replying :) 

I think I'll release a new version later this week (it took a bit more time), so then the real fix is implemented as well. 

Closing this for now :)

closed 2018-10-15 by fliphess