fliphess/chuangmi-720p-hack #20: chuangmi.jpg is not readable, because it is a RIFF-container

Staars, 2018-10-27:
Please choose another format, because WEBP is not exactly 100% compatible with all browsers and the current file suffix is wrong.

besides that:
Nice work!!

Staars, 2018-11-08:
I'm almost afraid to say, because it is really not so important, but the image is still not readable (for Safari and Firefox, but NOT Chrome, which does not even warn, that something is unusual). 

Maybe, we must update this file (?): 
https://github.com/fliphess/chuangmi-720p-hack/blob/master/sdcard/firmware/www/public/static/images/chuangmi.jpg

But much more important: Great update! Thank's a lot!

fliphess, 2018-11-10:
Thanks for your kind words and your patience :)

I'm obviously not that good at testing in multiple browsers :)
If you have a suggestion yourself for an image that works for all, let me know, so I can add it.... 

Otherwise I'll see if I can find one later this week and add it to the next release :)

Staars, 2018-11-11:
Don‘t worry, this is only a minor cosmetical glitch. You did add a correct jpg-file in the root folder of your repo.
But you did not change this: https://github.com/fliphess/chuangmi-720p-hack/blob/master/sdcard/firmware/www/public/static/images/chuangmi.jpg

You can do a quick check with the web view on github.com.

👍

fliphess, 2018-11-11:
Ah now I see, there is a second image :) 

I changed it in https://github.com/fliphess/chuangmi-720p-hack/blob/development/README.md
Now the readme uses the image from the public root, which is not the RIFF image anymore. 

Thanks :)

fliphess, 2018-11-18:
I'm reluctant to close the issue until you've confirmed, but in the latest release this should be fixed :D

Staars, 2018-11-18:
Yes, please do. 
I will test it ASAP, but by looking at your commit I think, it should be really solved now.

If I find the time, I will try to contribute a little bit of work, but I cannot promise anything atm.

Staars, 2018-11-18:
YESSSS!!!
It is now correct.

commit 8f421876 (2018-10-31): Replace image with a real jpg one :) (Fixes: #20)
commit 6ebc11c7 (2018-11-08): Merge changes from development into master prior to a new release (#21)

* Add curl --cacert to aliases to make using curl with https a bit easier
* Fix documentation for api endpoints
* Fix path mapping for init scripts to services for telnet and rtsp
* Use gm_lib from our own repo so we can add some changes and optimization
* Speed up the build by not unpacking the whole linux distribution but only the toolchain
* Rename samples to utils, use sources to define all src archive sources
* Remove env vars for pushing docker container from .travis.yaml
* Add a turbo button: Free memory using the  [... cut]
closed 2018-11-18 by Staars