fliphess/chuangmi-720p-hack #23: Support sleep mode

fliphess, 2018-11-18:
In the official firmware it seems possible to put the camera into sleep mode and wake it up again triggering certain actions.  (mentioned in jymbob/chuangmi-720-hack#4)

Although it does not seem easy to re-implement this functionality, it sounds like a great feature.

Staars, 2018-11-18:
The question is, what is the definition of sleep mode in this particular case.

I think, they (the camera vendors) mean stopping video streaming and recording. Or am I wrong here?

fliphess, 2018-11-20:
Stopping services as a way of sleep seems easy to implement, but I think i can reverse engineer the existing feature by triggering it from the app while running some debug tracers on the camera and implement something similar. 

On the other hand i noticed there is a [RTC controller on the board](https://fliphess.com/toolchain/Software/Docs/LowLevel_Userguide/GM8136_RTC_User_Guide_V1.0.pdf) which, if enabled could provide some kind of "deepsleep" mode over i2c.

fliphess, 2018-12-08:
Okay I'm giving up on the deepsleep: I've managed to manage the rtc clock over i2c and even bring the  camera in some kind of lowpower non-responsive state through some functionality in the kernel, but this would only be useful if it managed to become responsive again....... 

I'll create a no-camera, no lights, sleeplike feature instead in the next few weeks :)

fliphess, 2019-01-27:
I'm moving all bugs and feature requests to their own project tracker. I feel I'm spending too much time on debugging configuration issues that are documented in the wiki so to focus on new features I'm closing this issue. 

I encourage everyone that runs into bugs to debug them theirselves and create a PR to fix the issue upstream.

closed 2019-01-27 by fliphess