fliphess/chuangmi-720p-hack #25: nano error: Error opening terminal: xterm-256color on macOs

Staars, 2018-11-18:
This is not a severe problem and already discussed here:
https://stackoverflow.com/questions/6804208/nano-error-error-opening-terminal-xterm-256color

For me it is fine to change the setting in the terminal app on the Mac to VT100, if I need to use nano over ssh.

I just wanted to mention it and maybe a short info in the Wiki could be added.

fliphess, 2018-11-19:
Ah! thanks for filing an issue :)

I think setting one the available term profiles in the /root/.profile so it will be set at login time should resolve this. 

If you want to set it for your SSH session after login, you can manually set it in your environment by executing:

```
export TERM="xterm-color"
```

or by adding it to your .profile: 

```
echo 'TERM="xterm-color"' >> /tmp/sd/firmware/root/.profile
echo 'TERM="xterm-color"' >> /root/.profile
```

I will add it to the new release asap.

fliphess, 2018-11-19:
For future reference: There are several available options for configuring your terminal. 
Available term configurations are:

```
ansi
linux
screen
vt100
vt102
vt200
vt220
xterm
xterm-color
xterm-xfree86
```

(Which I will add to the wiki as well)

Staars, 2018-11-19:
Thanks for the additional infos.

fliphess, 2018-11-23:
I added [this settings to the wiki](https://github.com/fliphess/chuangmi-720p-hack/wiki/Configuring-the-your-terminal-profile) and set a default configuration in the `.profile` for root.

Will close this when the newest release is out in the open :)

commit 6a954ec8 (2018-11-23): Set default TERM settings in /root/.profile (Fixes: #25)
closed 2018-11-23 by fliphess