fliphess/chuangmi-720p-hack #27: Allow change of width, height, fps etc in rtsp

fliphess, 2018-11-22:
Use the patch from: https://github.com/willthrom/rtspServer_gm to allow change of fps etc.

Staars, 2018-11-24:
I think that is a good direction.
What I would like to have, is the possibility to integrate the cam via Homebridge. This should already be possible with: homebridge-camera-ffmpeg
But this would mean a more or less re-encoding of the rtp stream. I'd rather would modify the rtspserver so that all encoding and computing is done on the cam.
In theory this could be started in `FFMPEG.prototype.handleStreamRequest = function(request)` in https://github.com/KhaosT/homebridge-camera-ffmpeg/blob/master/ffmpeg.js, where the stream is normally handed to FFMPEG.
But I have no idea, if this is technically possible on the cam and this topic is a bit above my skills in this area.

fliphess, 2018-11-25:
To what kind of size would you like to encode? Then I can do some testing :)

fliphess, 2018-11-25:
I compiled the rtspd that allowed to change the width, height and framerate, but there are some minor differences in the timing resulting in a glitchy stream, so i'm looking into merging the 2 without keeping the timing changes.

At this moment using the [rtspd from the willthrom repository](https://github.com/willthrom/rtspServer_gm) results in a stream quality that is much worse than with the one i'm using, even though their origin is the same source code from grain media / faraday technology so there must be some other changes resulting in this quality drop. 

I'm looking into what changes could've caused this change.

fliphess, 2018-11-25:
The hardware is very picky about which settings work: using another framerate than the supported 15fps will results in errors and a very unsmooth stream and changing the size/resolution I think would/should be possible but encounter the same issue: it is hard to find the right height/width as not all resolutions and combinations are working smoothly with the hardware in the chuangmi. 

I will try to find the best results possible and use those as the defaults and add a big fat do not change if you don't feel like wasting time on testing warning. 

In the ideal world we would find a way to use multiple sizes using a different endpoint, which is possible and already shown in the code how to initialize multiple streams. 

I'll test if i can add multiple encoder threads resulting in something like:
- /live/640x480
- /live/800x600
- /live/1280x7820 

But I'm not sure if the hardware supports such a mass conversion from and to so I'll do some tests to see if it's possible. 

(Multiple streams seems to be sort of possible using h264 though..... Not with mpeg, but this is not an issue for ffmpeg)

Staars, 2018-11-25:
Thanks for your investigations.
I have no experience with the above mentioned homebridge-plugin yet, but I assume, that the default-settings in the file 
https://github.com/KhaosT/homebridge-camera-ffmpeg/blob/master/ ffmpeg.js could be a good starting point.

fliphess, 2018-11-26:
As far as I see, using homebridge-camera-ffmpeg should already be possible with some tweaks in the configuration of the plugin. 

If you use the following settings:

```
fps: 15
height: 720
width: 1280
```
I think you're already there :) 

If possible, I would recommend doing the conversion to other formats off-camera where possible as there is probably only enough resources available to do a single conversion.

closed 2018-12-08 by fliphess

fliphess, 2018-12-08:
It's now possible to change the framerate, size and fps settings by giving commandline arguments to the rtspd binary. Let me know how it works and if it's compatible with the homekit plugins you wanted to use :) 

If it works, feel free to write some integrations docs for homekit if you'd like

Staars, 2018-12-08:
I hope to find some time in the next week and would like to report back. (but this time thing might be a problem ;)

commit a3e8befa (2018-11-25): Use commandline arguments for setting framerate,height,width,bitrate etc (Fixes: #27)