fliphess/chuangmi-720p-hack #29: Write integration documentation

fliphess, 2018-11-25:
Create wiki articles to explain the following integrations: 

- home assistant 
- homebridge 
- zoneminder 
- domoticz

etc.

markusdavidhu, 2018-12-10:
Hi! Can you help me how can I integrate the camera to Home Assistant with MQTT control?

fliphess, 2018-12-11:
I'd love to provide the instructions, but my home assistant is running a very old version, so I'm not sure if it's still as accurate as it should be. 

I'm using zoneminder myself as this has the option to proxy a single rtsp connection from the camera to multiple clients, which is a bit more lightweight for the camera itself. 

Without it, you can use the [ffmpeg component](https://www.home-assistant.io/components/camera.ffmpeg/) to let home assistant take care of the connections to the camera. 

For me what worked, is this:
```
- platform: ffmpeg
  name: Chuangmi
  input: -rtsp_transport tcp -i rtsp://camera:554/live/ch00_0
```

Let me know if it still works and what your experiences are... Feel free to write some integration docs ;)

markusdavidhu, 2018-12-11:
Hi! Thank you, the video is working fine, but I can't get to work the MQTT control in Home Assistant. I can see the camera in the log, but I don't know how to integrate the control buttons. (Camera is connecting to the broker, but I don't know how to control it from Home Assistant)

fliphess, 2018-12-11:
How to control your camera in home assistant is a bit up to you as you should decide for yourself which functionality you want to make controllable through home automation. 

For example, you can choose to create buttons to enable and disable the leds or the rtspd service using [mqtt switches](https://www.home-assistant.io/components/switch.mqtt/): 

```
switch:
  # Change blue led
  - platform: mqtt
    name: "Camera1 Blue Led"
    icon: "mdi:led-on"
    state_topic: "home/camera1/leds/blue"
    command_topic: "home/camera1/leds/blue/set"
    payload_on: "on"
    payload_off: "off"

  # Change rtsp service on/off
  - platform: mqtt
    name: "Camera1 MJPEG RTSP Server"
    icon: "mdi:cctv"
    state_topic: "home/camera1/rtsp"
    command_topic: "home/camera1/rtsp/set"
    payload_on: "on"
    payload_off: "off"
```

(Untested but mostly copied from my own hass)

... Or create [sensors](https://www.home-assistant.io/components/sensor.mqtt/) or [binary sensors](https://www.home-assistant.io/components/binary_sensor.mqtt/) for topics/outputs that are not controllable but only send status updates... (binary_sensor if the output is 1/0 (on/off)  and sensors for anything else):

```
sensor:
  - platform: mqtt
    name: Camera1 Light Sensor
    state_topic: "home/camera1/brightness"
    unit_of_measurement: "%"
    icon: "mdi:brightness"
```

The output of system_status is:
```
{"uptime":" 20:52:54 up  4:06,  1 users,  load average: 21.25, 21.24, 21.23","ssid":"gnoffel","bitrate":"72.2 Mb/s","signal_level":"-57","link_quality":"87%","noise_level":"0"}
```
Using templates you can split each part to it's own sensor:
```
sensor:
  - platform: mqtt
    name: Camera1 Noise Level
    state_topic: "home/camera1/system"
    unit_of_measurement: "%"
    value_template: '{{ value_json.noise_level.replace("%", "") | int }}'
```

Have a look which commands are available for control (switches):
https://github.com/fliphess/chuangmi-720p-hack/blob/master/sdcard/firmware/scripts/mqtt-control.sh

And for status updates (mqtt sensors and or mqtt binary sensors):
https://github.com/fliphess/chuangmi-720p-hack/blob/master/sdcard/firmware/scripts/mqtt-interval.sh

I would not recommend to map each functionality and part of the output the camera spits out over mqtt to sensors/switches etc but first determine what you want to do and only create configuration to do that. Home assistant changes a lot and having to control all your configuration at every update is a bit of a pita so keep your config as small as possible and only create what you need :)

fliphess, 2018-12-11:
Add the load:

```
  - platform: mqtt
    name: Camera1 Load 1min
    state_topic: "home/camera1/system"
    unit_of_measurement: ''
    value_template: '{{ value_json.uptime | regex_findall_index("[\:|,]\ ([0-9\.]+)", -3) | float }}'

  - platform: mqtt
    name: Camera1 Load 5min
    state_topic: "home/camera1/system"
    unit_of_measurement: ''
    value_template: '{{ value_json.uptime | regex_findall_index("[\:|,]\ ([0-9\.]+)", -2) | float }}'

  - platform: mqtt
    name: Camera1 Load 15min
    state_topic: "home/camera1/system"
    unit_of_measurement: ''
    value_template: '{{ value_json.uptime | regex_findall_index("[\:|,]\ ([0-9\.]+)", -1) | float }}'
```

fliphess, 2018-12-11:
To get to know which functionality is available i suggest testing using the mosquitto utilities: 
using `mosquitto_pub` you can publish messages to the broker and see how the camera responds. 

I will add some logging for the MQTT service so you can monitor what is happening while testing.

If you are on a windows machine and installing `mosquitto_pub` is not an easy option: you can use the mosquitto utilities on the camera itself as well, as the developers of the original firmware were kind enough and left it there for us to use :)

fliphess, 2018-12-15:
Hey @markusdavidhu, did you manage to get it working or is a bug in the mqtt scripts blocking you from setting up MQTT?

Love to hear if you made some progress....

markusdavidhu, 2018-12-17:
Hey @fliphess , thanks for your example, I can control the camera via MQTT in HA, but sometimes the control is not working, or the state not changing, but after a restart it works normally again.

fliphess, 2018-12-18:
I'm suspecting the mqtt connection of the listener to be dead/disconnected when it's at its busiest and the mosquitto_sub utility not reconnecting after disconnection leaving the daemon in limbo. 

If the camera is not responding, is it completely stuck and not responding to any mqtt messages at all or does it "come back" after an amount of time?

When you restart, do you mean you restart the whole camera? (IE: power cycling) or restarting the MQTT script? 

In case you restart the camera, can you verify if rather than rebooting restarting the service suffices? (`/etc/init/S99mqtt-control.sh restart`)

To cross some things off:
- Can you post the output of `ps -ef | grep -vE '\[.+\]'` on the command line?
- The output of `uptime` as well

It's good to know that the camera is (by design) constantly overloaded. It's doing way to much for the cpu to handle, so it might be because of running too much things at once :) 

I'm having some days off soon and I plan to fix some things to make it more robust and in the less near future rewrite the mqtt control and interval daemon to C.

markusdavidhu, 2018-12-21:
Hi @fliphess! Sorry for my late response. The camera is not responding to anything, It is completely suck, and I don't know that is it come back after an amount of time. Until now I restarted the whole system from the web UI, but i think that the mqtt service reboot is enough to come back. I will try.

fliphess, 2018-12-22:
Hey @markusdavidhu ! 

If restarting the MQTT services is enough to get going again, I think it has to something do with the MQTT connection from the camera to the broker getting closing and not being reconnected. I'll have to do some research to confirm, but I'm sure we can easily fix this. 

As the end of December is not a good time to hurry things, I think it will take another week before I can fix it.

fliphess, 2018-12-23:
Eh shuffling issues was not the best thing to do.... I've created an issue for the MQTT unresponsiveness, leaving this open for integration with home automation suites.

fliphess, 2019-01-20:
Documented most of the comments: Closing this as the issue got hijacked :D

closed 2019-01-20 by fliphess