fliphess/chuangmi-720p-hack #51: Measure light intensity from sensor using proc node and switch night mode

fliphess, 2018-12-11:
Create a little daemon that switches night mode on and off when the light intensity is below a certain value for a measured amount of time. 

In other words: Create a more useful `ir_sample` that manages the ir_led as well.

epalzeolithe, 2019-01-20:
Hi, il confirm, it's needed ;)

fliphess, 2019-01-20:
Thanks :) I'm working on it, but lower your expectations as still I'm learning C and coding the utilities while learning (one step forwards, 2 steps back). This is making my progress not always that fast ;)

If you have an urgent need for this functionality I'd recommend using the EV and IR values from `nightmode -j` (use the values that are send to the MQTT broker every N seconds) from mqtt-interval.sh.

Using a home automation suite like home assistant you can easily add an automation for changing the IR Led, Nightmode and IR Cut based on incoming MQTT messages.

fliphess, 2019-01-25:
@epalzeolithe I've created a first version that works similar to the `ir_sample` utilty. 
I'm working on and towards a release this weekend, hope it does what you are looking for :)

epalzeolithe, 2019-01-27:
Hi,http 500 error, i cannot command the camera, and the colors are ugly, very strange