fliphess/chuangmi-720p-hack #58: [WEB UI] - Error retrieving NVRAM value for key ir_led

chris777c, 2018-12-22:
Hello,

Maybe you know that but I can't see any issue talk about that. When I navigate to the WEB UI, in tab Setting, He show me, this message : Error retrieving NVRAM value for key ir_led.

You doing a very good job for the camera

fliphess, 2018-12-23:
Can you check what the `ir_led` value is in NVRAM settings on the same page or using `/api/nvram/state`? 

If it's not set, it's possible it got deleted or corrupted. Try adding it on the command line:

```
nvram set ir_led=off
```

Turning on the led once should do the same thing. If that doesn't help, let me know, it then might be something in the scripts....

fliphess, 2018-12-23:
You could do that to workaround the error, but I think a restart does the same.

It's more interesting how the settings disappeared or the error is caused. 
Did you delete values from nvram yourself manually or did a buggy script caused it? 

The `ir_cut` and `ir_led` settings are [set at boot time](https://github.com/fliphess/chuangmi-720p-hack/blob/master/sdcard/firmware/scripts/.boot.sh#L41).... 

If you receive that errors without changing the nvram, I expect some script errors at startup. Can you check the log files if you see anything errorish?

chris777c, 2018-12-23:
I add ir_cut in nvram, this conf stay after restart ==> good thing

You see below, the log boot 

```
*** Stopping restartd Restartd daemon ...
PWM initialization failed
Failed to open /dev/ftpwmtmr010
PWM initialization failed
Failed to open /dev/ftpwmtmr010
*** Starting auto_night_mode Auto Night Mode... 
*** Starting rtspd RTSP server... 
*** Starting lighttpd web server... 
*** Starting ntpd NTP Daemon... 
*** Stopping crond Cron Daemon ...
*** Starting dropbear SSH server... 
*** Copying previous key to /etc/dropbear... OK
*** Stopping telnetd Telnet Daemon ...
*** Waiting for ping reply from 192.168.* for 10 times..... OK
*** Configuring NVRAM
*** Mounting /tmp/sd/firmware/bin on /tmp/sd/ft... 
*** Executing /mnt/data/test/boot.sh... 
*** Disabling miio_ota... 
*** Disabling OTA... 
*** Disabling logmi.sh... 
*** Disabling miio_smb... 
*** Disabling miot_qrcode.sh... 
*** Disabling miot_devicekit... 
*** Disabling miio_client_helper_nomqtt.sh... 
*** Disabling miio_client... 
*** Disabling miio_avstreamer... 
*** Disabling Cloud... 
Configure WiFi
*** Configuring WIFI... 
Failed to load lowcase.dat, will use lame ASCII-only case sensitivity rules
*** Setting Samba root password... Failed to load upcase.dat, will use lame ASCII-only case sensitivity rules
Password for 'root' changed
*** Setting root password... 
*** Appending /tmp/sd/firmware/etc/hosts to /etc/hosts
*** Configuring new /etc/hosts file... 
*** Setting hostname... 
*** Configure time zone... 
*** Replacing restartd startup script and config with our own version
*** Mounting /root from sd card
*** Making /etc writable
*** Starting logging... OK
*** Stopping logging... OK
*** Enabling logging

  ENABLE_RTSP    = 1
  ENABLE_LOGGING = 1
  ENABLE_SAMBA   = 0
  ENABLE_FTPD    = 0
  ENABLE_HTTPD   = 1
  ENABLE_SSHD    = 1
  ENABLE_TELNETD = 0
  DISABLE_OTA    = 1
  DISABLE_CLOUD  = 1
  NTP_SERVER     = 192.168.*
  TIMEZONE       = *
  WIFI_PASS      = *
  WIFI_SSID      = *
  HOSTNAME       = *

Chuangmi 720P hack configuration:

Running Chuangmi 720P hack
```

chris777c, 2018-12-24:
Since I had add myself ir_cut and ir_led in nvram and it's stay in nvram even thougt restart , I can't still occur this bug.

fliphess, 2018-12-24:
Did you delete the `ir_*` settings from nvram yourself or did some script remove the settings from nvram?

chris777c, 2018-12-29:
No I didn't remove ir_* myself and I didn't have any script who do that ? I had just copy your release in sdcard and start webcam.

fliphess, 2018-12-31:
That is interesting :) Let's leave this issue open for a bit longer so I can figure out if it happens more often :)

chris777c, 2019-01-01:
I copied the last version V 0.959 on my sd card and the bug is return (I try to disable auto night mode and  the bug is always here)

fliphess, 2019-01-04:
Strange.... I haven't been able to reproduce it so far, but I'll check with another camera later this week to see if it appears then.  

I'm still not fully it's not a bug in the functions.sh where `auto_night_mode` is involved.... I'll look into it.

(I'm quite busy the next coming weeks so it'll take some time.)

fliphess, 2019-01-05:
It was indeed a bug at startup..... I've added a commit that should fix this issue in the next release :) Thanks for filing the issue! 👍

closed 2019-01-13 by fliphess

fliphess, 2019-01-26:
Sad to say the error was still there.... I was too quick in closing this issue. I will release a fix in the next release (v0.963)

commit 73c3315a (2019-01-05): Set ir_cut, ir_led, blue_led and yellow_led correctly at boot (Fixes: #58)
commit db039b13 (2019-01-13): Merge development into master prior to release v0.960 (#79)

* Remove syntax highlight from vimrc
* Bink blue led when motion is detected
* Make rtspd write the location of the image to a file when a snapshot is created
* Add some of our own libs, rename rtspd to rtsp-server for consistency in makefile
* Make a clear separation between our own cruft and third party stuff
* Remove runas: su is available and we don't use it for now
* Remove samples: We're not using them and they're both in the toolchain and the chm file.
* Write frame data to file on command and on motion
* Fix zlib cd error dur [... cut]