fliphess/chuangmi-720p-hack #59: [Q] - How to start and stop by API camera ?

chris777c, 2018-12-22:
Hello,

I would like start and stop camera by API (stop image, led, ... ) , how to do that ?

Other question nothing see with this message : How to give you money for your job ?

fliphess, 2018-12-22:
Hey @chris777c! 

When you type start and stop, do you mean the camera stream (the rtsp service) or the camera? (shutdown/restart) ?

To give an example about HTTP control: If you want to make an API call to turn on and off a blue led, you can do this using a tool that can do http requests. In this example i use curl, but wget, a browser, the rest component of home assistant all accomplish the same thing:

```
## Turn led on:
curl --user root http://camera01/api/led/blue_led/set/on

## Turn led off:
curl --user root http://camera01/api/led/blue_led/set/off
```

``` 
## RTSP on
curl --user root http://camera3/api/services/rtsp/start

## RTSP off
curl --user root http://camera3/api/services/rtsp/stop
```

Is this answering your question or did i misunderstand the question asked?

About giving: I don't think I would feel very comfortable to get paid or receive donations for my work on this project as to me this project is pretty much a way of relaxing after work. If I don't feel like it,  I don't work on it, which makes it a perfect project to be busy with while procrastinating other things :) 

If you insist on donating anyway, I'd  prefer see that you make a donation to a charitable organisation that can use it a lot better than me :) (For example https://www.sgtpeppersfriends.com/)

chris777c, 2018-12-22:
> When you type start and stop, do you mean the camera stream (the rtsp service) or the camera? (shutdown/restart) ?

I would like shutdown and restart camera.

chris777c, 2018-12-23:
OK I understand, good work for motion détection.

For me I use [image] solution with one motion detection by an another [image]. It's for thaht I ask to you for stop and start camera by API. Like that I start camera rec on xiaomi motion détection is up.

fliphess, 2018-12-23:
Ah I see what you mean :)  Do you use external motion detection devices? 
I'm working on a feature that can record a few seconds of video, which can be useful if a sensor detects motion. (#40)

chris777c, 2018-12-23:
I edit my previous post like that link work.

fliphess, 2018-12-24:
If you want to fully start and stop the camera, I think you should try control the power using another device (a sonoff or some other wifi controlled relay switch. 

The camera itself will never be able to turn itself on sadly as there is an egg in the way of the chicken :)

closed 2018-12-24 by fliphess