fliphess/chuangmi-720p-hack #83: Not booting from SDCARD

tomsteenbakkers, 2019-01-14:
@fliphess I installed the new version, but no changes with the previous version. The camera is booting and it is also recording, they are stored in the sd-card. When the blue light is on, I can't connect using SSH with the camera. Also browsing to the IP nummer has no result.

Is there a log-file where I can see where it goes wrong.

fliphess, 2019-01-14:
Hey @tomsteenbakkers! 

Did you reset using the `tf_recovery.img` image? If the downgrade succeeded it should be renamed to `tf_recovery.img.bak`......

After that, is SSH enabled? If not, enable it in the configuration file and then check whether the SSH daemon is accessible and you are able to connect. 
After that, Login and check the logs in `/tmp/sd/logs`. 
Hopefully there is some errors in the log file..... 

if not, there's a big chance you'll be able to connect using telnet, with or without authentication.

fliphess, 2019-01-14:
If the camera is recording on the sd card, what is the name of the directory the media files are stored in?

fliphess, 2019-01-15:
If the recording functionality is still active, i'm kind of sure the original firmware is still running and the hack is not being used. 

If this is the case you should be able to connect to telnet, which is activated in the tf_recovery image. Using telnet you should be able to run `cat /tmp/sd/log/ft_boot.log`. I'm wondering if there's an error in the boot process that is causing this.

tomsteenbakkers, 2019-01-15:
@fliphess I have access with telnet so the tf_recovery image did his work. The command cat `/tmp/sd/log/ft_boot.log` is not working there is no `/temp/sd` directory. Is there another location where log files are stored?

fliphess, 2019-01-15:
Hey @tomsteenbakkers 

`/tmp/sd` is the location where the sd card normally is mounted. 

The sd card not being mounted can have quite a lot of 

Can you post the output of `df -h`?  Let's see if there is a mounted sd card somewhere on the filesystem.

If no SD card is mounted at all, it could be because of a corrupt sd card or some formatting issues.

tomsteenbakkers, 2019-01-15:
I will try another SD card. To restore the firmware to the original image I used the same SD. 

here the content of `df -h`

```
Filesystem                Size      Used Available Use% Mounted on           
/dev/root                 8.6M      8.6M         0 100% /                    
devtmpfs                 29.9M         0     29.9M   0% /dev                 
tmpfs                    29.9M         0     29.9M   0% /dev/shm             
tmpfs                    29.9M    100.0K     29.8M   0% /tmp                 
tmpfs                    29.9M         0     29.9M   0% /run                 
tmpfs                    29.9M     80.0K     29.8M   0% /var                 
tmpfs                    29.9M         0     29.9M   0% /mnt/media           
/dev/mtdblock3            2.9M      1.3M      1.6M  44% /mnt/data            
/dev/mmcblk0p1            1.9G    153.6M      1.7G   8% /mnt/media/mmcblk0p1 
```

roboterm, 2019-01-15:
@tomsteenbaker did you formatted the sentence card on FAT32?
On a macOS system if the as card is smaller than 32gb it’s only possible to format a sd card on FAT32 via Terminal.

tomsteenbakkers, 2019-01-15:
> @tomsteenbakkers did you formatted the sd card on FAT32?
> On a macOS system if the as card is smaller than 32gb it’s only possible to format a sd card on FAT32 via Terminal.

Yep, I used a Windows PC to format the SD card.

fliphess, 2019-01-15:
For some reason the sd card is not mounted on `/tmp/sd`, which should be done in [`ft_boot.sh`](https://github.com/fliphess/chuangmi-720p-hack/blob/master/sdcard/ft/ft_boot.sh#L38)

It is mounted on `/mnt/media/mmcblk0p1`, but for some reason the mount in our injected boot script is not created. I'm wondering whether the ft_boot.sh script is run at all.....

Can you post the output of: 

```
cat /tmp/ft_mode
```

and:
```
ls /mnt/media/mmcblk0p1
```

And last, to check if its possible to mount the sd card to another location:
```
mkdir /tmp/sd
mount -t vfat /dev/mmcblk0p1 /tmp/sd
```

Hopefully this will show me what is going on, otherwise we have to dive deeper in the hack, which is executed in `/etc/init.d/rcS`, but we'll get there when we get there...

tomsteenbakkers, 2019-01-15:
Tried another SD card but with the same result. Then I tried to connect the camera with the Mi Home app but without any succes. Now, if I reboot the camera with or without the SD card it waits for a connection.

tomsteenbakkers, 2019-01-15:
Camera is working again and telnet is working. This is the output:

```
mijia-camera login: root                                          
# cat /tmp/ft_mode                                                
0                                                                 
# ls /mnt/media/mmcblk0p1                                         
MIJIA_RECORD_VIDEO  firmware            ft_config.ini             
config.cfg          ft                  log                       
# mkdir /tmp/sd                                                   
# mount -t vfat /dev/mmcblk0p1 /tmp/sd                            
#                                                                 
```

tomsteenbakkers, 2019-01-15:
I have downloaded your Fork, but when looking online it is not in de `SDcard` directory. Where can I download that file?

fliphess, 2019-01-15:
It is created at build time from the `Makefile`, you can create it manually using: 

```
cd <repo>
tar -cf sdcard/manufacture.bin manufacture/test_drv
```

Or you can unpack the tarbal or zip from the [releases page](https://github.com/fliphess/chuangmi-720p-hack/releases) and get it from there.

tomsteenbakkers, 2019-01-15:
Donwloaded the source and the hack is working now. Look very good, lots of option! Thanks

fliphess, 2019-01-15:
Ah nice! Good to hear :)

closed 2019-01-15 by fliphess