fliphess/chuangmi-720p-hack #86: Upper the FPS limit to 30fps instead of 15fps

fliphess, 2019-01-16:
The hard limit in gmlib.cfg is 15fps, but it would be nice to go to 30fps if the hardware supports it.

fliphess, 2019-01-16:
See https://github.com/ghoost82/mijia-720p-hack/issues/21 by @Nuker1773 for more information.

fliphess, 2019-01-19:
@Nuker1773: Changing the framerate to 30 FPS appears to be easy, although I cannot guarantee the camera does not burn down. 

I notice a small increase in power usage and as I use a variable bitrate the framerate never gets to 30 but stays around 25 fps, which is good enough for me. 

Changing the framerate +15fps will be added in the next release.

Nuker1773, 2019-01-20:
You are awesome!

Nuker1773, 2019-01-21:
sounds great. Is there any temperature sensor in the device?

fliphess, 2019-01-21:
As far as i know there is no temp sensor on-board..... That would have been nice :)

Nuker1773, 2019-01-26:
Hi fliphess, I saw the new release with higher fps possibilitys. How to Set up? When using more the 15 fps in the rtsp config section I get Invalid Input in the rtsp.log

fliphess, 2019-01-26:
Ehm in `/etc/init/S99rtsp` the settings for a bigger framerate should be set: 

```
## Set max FPS to 30 if fps > 15
if [ "$RTSP_FRAMERATE" -gt 15 ]
then
    echo w sen_fps 30 > /proc/isp328/command
fi
```

Can you check what the output is if you run:

```
echo r sen_fps > /proc/isp328/command
cat /proc/isp328/command
```

And if that shows: `sensor fps=15`, try setting it to 30:

```
echo w sen_fps 30 > /proc/isp328/command
```

Let me know if that suffices.

Nuker1773, 2019-01-26:
it was showing fps=15. After setting it to 30 its output shows fps=30 but the RTSP Video stays at 15 fps. any idea? BTW. Thanks for your work!

fliphess, 2019-01-26:
The framerate stays below the 30, especially if you set a variable bitrate (bitrate mode). If you use a constant framerate it should be around 20, 25. 

It depends a bit on the framerate, the bitrate and the bitrate mode, but the camera is not always equipped to do so much encoding in so little time. 

It might help to try some different encodings or bitrates and bitrates modes, but don't expect too much.

When I set the mode to CBR, the bitrate around 2048 and the bitrate mode to constant, turn off MQTT and the lighttpd server, I can squeeze it to 25 frames FPS if it's not too dark in the room.... 

To be honest, I'm quite impressed of that this shitty little egg camera is capable of such a thing and I don't expect it to get any better than this.

fliphess, 2019-01-26:
If you reboot the camera and re execute the commands to check the framerate, is the sensor framerate set correctly or is it at 15 fps again? (wondering if there's a bug in the rtsp init script)

```
echo r sen_fps > /proc/isp328/command
cat /proc/isp328/command
```

Nuker1773, 2019-01-26:
Okay I will try some Settings.. right now its very dark. But you are correct, after reboot its shows sensors fps=15 again.

fliphess, 2019-01-26:
That is not what it should be, I think it's not set correctly set up in the rtsp init script. 

Just to make sure: You already changed the settings in your `config.cfg` from `RTSP_FRAMERATE=15` to `RTSP_FRAMERATE=30`? 

The default setting remains at 15 fps to keep it as stable as possible in general.

fliphess, 2019-01-27:
I'm moving all bugs and feature requests to their own project tracker. I feel I'm spending too much time on debugging configuration issues that are documented in the wiki so to focus on new features I'm closing this issue. 

I encourage everyone that runs into bugs to debug them theirselves and create a PR to fix the issue upstream.

commit be80324c (2019-01-19): Allow a higher framerate than 15FPS (Fixes: #86)
commit a5c56fd1 (2019-01-25): Release v0.961

Containing:

    * Add a script to restore the state of the settings after a reboot
    * Add a first version of a home brewed auto night mode switcher
    * Add sliders for brightness contrast etc to settings page
    * Enable cron by default (again)
    * Add an endpoint to request a image directly
    * Add phpinfo to api docs and add some ocd
    * Allow a higher framerate than 15FPS (Fixes: #86)
    * Refactor mqtt scripts

    * Add an alias to exit the shell after reboot to prevent hanging ssh sessions while testing
    * Make recording errors a bit more user friendly, f [... cut]
closed 2019-01-27 by fliphess