fliphess/chuangmi-720p-hack #88: Downgrade not working

epalzeolithe, 2019-01-19:
Hello i did use the tf_image to downgrade

I had the .bak file so that i'm sure it's downgraded

I've from mi-app, the 3.3.6_2017092810 version.

Nevertheless, never it boots the hack files on sd-card

Does anyone have this problem ?

tomsteenbakkers, 2019-01-19:
Check In the directory on the SD card of a file called is missing: manufacture.bin.

It's essential in the chain of scripts that make this hack possible....

If not then get the correct content using this link:

https://github.com/fliphess/chuangmi-720p-hack/releases

epalzeolithe, 2019-01-19:
[log.zip](https://github.com/fliphess/chuangmi-720p-hack/files/2775850/log.zip)

Ok i found the issue. I was using a 64GB SDHC EVO SAMSUNG card. It dont detect the hack on boot.

I used an old OEM SD card, and now it boots.

But i dont connect to my wifi. After orange for 2 minutes, The LED is blue, no blinking.

Here's enclosed the log. I'm looking for the issue

Without the sd card, it works in mi-app

fliphess, 2019-01-19:
Hey @epalzeolithe, 

Looking at the logging, you have some windows characters in the `config.cfg` file that cause issues while booting:

```
/tmp/sd/firmware/etc/init/S01logging: /tmp/sd/config.cfg: line 6: ^M: not found
/tmp/sd/firmware/etc/init/S01logging: /tmp/sd/config.cfg: line 8: ^M: not found
/tmp/sd/firmware/etc/init/S01logging: /tmp/sd/config.cfg: line 11: ^M: not found
/tmp/sd/firmware/etc/init/S01logging: /tmp/sd/config.cfg: line 14: ^M: not found
```
Can you try to edit the `config.cfg` in another editor?

epalzeolithe, 2019-01-20:
Working :) 

i dont know why bu credentials in the config.cfg are not working so i used the miapp after reset

Now configuring for homebridge :)

Thanks ==

epalzeolithe, 2019-01-20:
Working perfect :) DONT USE brand NEW SDHC card !!!

closed 2019-01-20 by epalzeolithe

fliphess, 2019-01-20:
Thanks, I'll add it to the troubleshooting guide :) 👍