fliphess/chuangmi-720p-hack #92: Homebridge config and http URL for direct still image access

epalzeolithe, 2019-01-20:
Hello

i configured the camera-ffmpeg homebridge plugin with this JSON : 
{
  "platform":"Camera-ffmpeg",
  "cameras":[
     {
        "name":"Camera 1",
        "videoConfig":{
           "source":"-rtsp_transport tcp -re -i rtsp://stream:bJ2xnahtCgninraelmI@192.168.1.10:554/live/ch00_0",
           "stillImageSource":"-rtsp_transport tcp -re -i rtsp://stream:bJ2xnahtCgninraelmI@192.168.3.1.10/live/ch00_0 -vframes 1 -r 1",
           "maxStreams":2,
           "maxWidth": 720,
           "maxHeight": 480,
           "maxFPS":15,
           "vcodec": "h264_omx",
           "debug": true
        }
     }
  ]
}

i used the rtsp stream to have access to a still image.

Can we access a still image directly from http ? When i read the the API i've understood i first need to make a snapshot then request it 

Thanks in advance !!!!

epalzeolithe, 2019-01-20:
I found a solution on the homebridge side

i made a fork : `https://github.com/epalzeolithe/homebridge-camera-ffmpeg`

we need to pass to ffmpeg a duration limit of the stream to 5 seconds, instead of 1s hardcoded. It lets ffmpeg to find headers in rtsp stream then do the snapshot, otherwise, the stream is closed before making anysnapshot

Perhaps it could be also fixed in the firmware, by setting some parameter in rtsp server

By the way, thanks a lot fliphess for your work ;è)

fliphess, 2019-01-20:
@epalzeolithe the camera is not using ffmpeg anymore for creating a snapshot but it creates a snapshot using the C libraries created by the manufacturer.  

Using ffmpeg off-camera to create snapshots sounds like a great idea, but on the camera it was a bit too heavy which resulted in sudden reboots.

If you have a nicely working setup with homebridge and ffmpeg, It would make me very happy if you could write some words for the implementation docs on how to use homebridge with this hack so I can put it in the wiki :) (hint hint) :D

epalzeolithe, 2019-01-20:
HEre's the tuto you can add to the doc

To add this camera to homebridge then to homekit, you need
1) HOMEBRIDGE installed somewhere (raspberry PI for instance)
2) then install the plugin : homebridge-camera-ffmpeg (https://github.com/KhaosT/homebridge-camera-ffmpeg)
3) then install the following JSON in the platform category
```
{
  "platform":"Camera-ffmpeg",
  "cameras":[
     {
        "name":"Camera 1",
        "videoConfig":{
           "source":"-rtsp_transport tcp -re -i rtsp://stream:bJ2xnahtCgninraelmI@192.168.x.x:554/live/ch00_0",
           "stillImageSource":"-rtsp_transport tcp -re -i rtsp://stream:bJ2xnahtCgninraelmI@192.168.x.x:554/live/ch00_0 -vframes 1",
           "maxStreams":2,
           "maxWidth": 720,
           "maxHeight": 480,
           "maxFPS":15,
           "vflip": true,
           "hflip": true,
           "vcodec": "h264_omx",
           "debug": true
        }
     }
  ]
}
```

4) to fix the snapshot, u need to manually edit the ffmpeg.js of the plugin to modify the timeout of ffmpeg request for snapshot from 1s to 5s (until it's implemented in the plugin)

epalzeolithe, 2019-01-20:
> @epalzeolithe the camera is not using ffmpeg anymore for creating a snapshot but it creates a snapshot using the C libraries created by the manufacturer.
>
> Using ffmpeg off-camera to create snapshots [...]

Good news, so you could in the future add the feature of direct snapshot as jpeg format from HTTP request. That will be perfect and will reduce the load on the homebridge server :) I look forward :)

fliphess, 2019-01-20:
Thanks! :) 👍 I've added your instructions to the wiki: https://github.com/fliphess/chuangmi-720p-hack/wiki/Integrating-the-cam-in-your-home-automation#using-home-bridge

fliphess, 2019-01-20:
One more question about the direct access link: Would it work for the homebridge plugin if the http request does not serve the image itself but instead redirects to the direct image location? 

Serving the image file over php is possible, but in the same time the file is already accessible using another public link, so if i could redirect using a http status code `301` or `302`, that would make things much more lightweight :)

epalzeolithe, 2019-01-20:
> One more question about the direct access link: Would it work for the homebridge plugin if the http request does not serve the image itself but instead redirects to the direct image location? [...]

For homebridge plugin, we need a direct http request which serves the jpeg file. The http url is passed directly to ffmpeg.

closed 2019-01-20 by fliphess

fliphess, 2019-01-25:
@epalzeolithe I've created a direct http link in v0.961

commit ce2e76d2 (2019-01-20): Add a new endpoint to download an image (Fixes: #92 and #94)