fliphess/chuangmi-720p-hack #93: Remote RTSP

alcestneige, 2019-01-20:
Please is it enough to open 554 port on router for rtsp or is there also something else to configure before connecting from different location to the camera. Thanks!

fliphess, 2019-01-20:
Please do not connect the camera from the internet! Read the [Security Considerations](https://github.com/fliphess/chuangmi-720p-hack/wiki/Security-Considerations) 🤕

closed 2019-01-20 by fliphess

fliphess, 2019-01-20:
Still, it's not safe to put on the webs, so if you want to access it remotely I recommend using a vpn solution, a secure rtsp proxy or SSH tunnel to your home server or something.

I use zoneminder myself that is accessible through an openvpn connection, but this requires some additional hardware and technical know how to setup.

Dropbear (SSH) is from a recent version so if you are able to create a port forward for only port 22 to the camera and then access the webserver and rtsp stream through an SSH tunnel that could work as well.

Opening up the whole camera (including web/rtspd) to the internet is a bad idea as it's a matter of time before someone uses it to explore your network or abuse the camera for remote attacks coming from your ip.

fliphess, 2019-01-23:
Hey @alcestneige, 

I've been browsing a bit and what i found is [kerberos-io](https://doc.kerberos.io/) 
It looks like a nice lightweight camera proxy that you can install on a raspberry pi. 

Using this tool you can connect your kerberos to your camera over wifi and expose kerberos to the internet watch it from work. It's not as heavy as using zoneminder or other camera recording solutions and it is a lot safer to access from the web :) Have a look at it, it might be the missing link in the chain :)

alcestneige, 2019-01-23:
Hello and thank you for the info, I will look into that, sounds good to me. Thanks again for your help and all best to you!