fliphess/chuangmi-720p-hack #95: Unable to boot

roboterm, 2019-01-21:
Hey!
I got some issues using your files. 
I m able to hack the cam when using the instructions of @jymbob and the files of @ghoost82. 
But with your files it is not possible to reach the goal. 
That is very strange since you forked from there (correct me if I m wrong). Any idea what what could be the reason?

But at the moment the hack is a bit useless since the night mode doesn’t work at all. 
My workaround is to turn the night mode always on via Mii Home app and then boot from sd card so that the night mode works as intended.

fliphess, 2019-01-21:
Hey @roboterm! I'm almost 100% sure it's not useless, but a pebkac :) 

Can you zip the log files on your sdcard and upload them in this issue? Than I can have a look what is going on and why it is not working. 

About the night mode: You can choose either nightmode or auto night mode, but not both at the same time. If you have `AUTO_NIGHT_MODE=1`, i recommend turning it off so you be able to switch the night mode and IR led manually.

fliphess, 2019-01-21:
Also, good to know: If the miihome app is working, the hack is not active. 
If you are able to switch the night mode using the miihome app, the cloud suite is still running, which should be disabled when using rtsp.

roboterm, 2019-01-21:
Don’t get me wrong:
I ejected the sdcard after unplugging the cam from power consumption. Than I set nightmode always on in Mii Home app. Unplugged it again from power, inserted the sdcard and powered on again. The result is a working RTSP stream, also viewable in Apple's Home app with nightmode always on. 
I choosed that because the Auto night mode didn’t work properly.

fliphess, 2019-01-23:
Hey! Let me know when you have some logs uploaded, so I can have a look what is going on :)

I still don't know what is not working and how it relates to the hack rather than the installation, but hopefully the logs give me some more info.

fliphess, 2019-01-23:
In case you need some instructions: I've created [this article](https://github.com/fliphess/chuangmi-720p-hack/wiki/How-to-tar-or-zip-the-logs-when-filing-an-issue) to get you going.

roboterm, 2019-01-26:
It‘s working now. Thanks for your help!

closed 2019-01-26 by roboterm

fliphess, 2019-01-26:
Great! Thanks for letting me know :)