fliphess/chuangmi-720p-hack #102: settings.html shows "Error retrieving NVRAM value for key ir_led: Not found "

fliphess, 2019-01-26:
As read in #98, in some cases the `Error retrieving NVRAM value for key ir_led: Not found` appears when loading the `settings.html` page.

Strange enough the infra red led does not use NV ram to store state anymore.

fliphess, 2019-01-26:
As said in #98 

*I would love to know from which api endpoint the error is coming. Could you please delete your browser cache to be sure it's not a cache issue and reproduce the error in your browser with the developer tools open?

In the network tab you can see which request to /api/something this error is coming from.*

@tomsteenbakkers: hopefully you can retrieve the endpoint when the error appears so I can find out where it's coming from :)

fliphess, 2019-01-26:
Muh! I've found it already. Although the mechanism changed, the state was still retrieved the old way.

tomsteenbakkers, 2019-01-26:
👍🏻👍🏻

fliphess, 2019-01-26:
Love the "Fixed" lable already :)

commit 0cbcf96d (2019-01-26): Retrieve the state of the irled from the IR_Led class. (Fixes: #102)

Retrieve the state of the infra red led through it's own function instead of the
function that now only works for the blue and yellow led.
closed 2019-01-26 by fliphess