fliphess/chuangmi-720p-hack #99: Special Chars not allowed in ROOT_PASSWORD (Config Editor)

GSanCap, 2019-01-26:
It's only 3 questions, and if they're not relevant you can wipe them, but please describe as much as possible to make it easy for me to understand what is going on.

See https://github.com/fliphess/chuangmi-720p-hack/wiki/How-to-tar-or-zip-the-logs-when-filing-an-issue 

**What are you trying to do but fails due to a bug? What is the expected result?**

The Editor must accept the password as valid

**Do you see any errors? If so, please copy paste the error message, or add a zip or tarbal containing the log files. Use back-ticks to make it readable please :)**

"Invalid input in configuration content: ROOT_PASSWORD="Zju4igMyP@5s!" (line: 28)"

**How can I reproduce the error?**

Adding this password to ROOT_PASSWORD line in the config.cfg and the try to edit other value using the Built-In Web Based Config Editor

fliphess, 2019-01-26:
Thanks for filing the issue, I would prefer to adjust the config editors input escaping, but as the config.cfg is a sources environment file, there will always be some special chars that are not allowed to be used. 

I'll have the config editor check for specials chars and return an error and add some lines to the documentation to clarify.

fliphess, 2019-01-26:
I've added a line to the trouble shooting guide (https://github.com/fliphess/chuangmi-720p-hack/wiki/Troubleshooting#general-tips). 

I will add some additional input validation to prevent this soon!

fliphess, 2019-01-26:
I've added `@$&%()` to the regex of allowed chars in the configuration file. exclamation marks, quotes backticks and semicolon will sadly not work.

commit ad39c81b (2019-01-27): Add chars to allowed regex in configuration file (Fixes: #99)
closed 2019-01-26 by fliphess